volumio/Volumio2 #1956: Lost volumio.log with update to 2.773

kermitnc, 2020-05-06:
Once again the playback died while skipping to a different song using tidal.

I tried therefore to search the volumio.log for some reasons but the file volumio.log is missing.
Afterwards I created the file volumio.log but no entries at all. After a reboot, the created file is gone.

volumio, 2020-05-07:
volumio.log has been removed.

you can access full log by doing: sudo journalct -f

ashthespy, 2020-05-07:
This was done in https://github.com/volumio/Volumio2/pull/1951 as it's redundant as it's not saved across reboots. 

tl;dr `cat volumio.log` == `journalctl -o cat -u volumio.service` 

PS: I normally add an alias:
```bash
echo -e "alias vollogs='journalctl -f -o cat'" | sudo tee -a /etc/bash.bashrc
```

closed 2021-02-13 by ashthespy