volumio/Volumio2 #1965: Playlist options are broken

jdwoody, 2020-05-14:
Testing the latest Volumio release, 2.773, in VirtualBox before flashing RPis and several playlist functions do not appear to operate correctly:
- Started with a clean install from volumio-2.729-2020-03-18-x86.img and ran the updater
- Added a source, which populated the library
- Created a playlist
- From the playlist menu, both the Play and Add to Queue fail. Root cause appears to be looking in /mnt for the playlist information instead of /data. Clear and Play operates correctly.

xipmix, 2020-06-02:
I can reproduce this for 'Play' and I agree with the diagnosis
```
Jun 02 11:05:16 volumio volumio[899]: info: CoreCommandRouter::volumioAddQueueItems
Jun 02 11:05:16 volumio volumio[899]: info: CoreStateMachine::addQueueItems
Jun 02 11:05:16 volumio volumio[899]: info: CorePlayQueue::addQueueItems
Jun 02 11:05:16 volumio volumio[899]: info: Adding Item to queue: playlists/test
Jun 02 11:05:16 volumio volumio[899]: info: Exploding uri playlists/test in service mpd
Jun 02 11:05:16 volumio volumio[899]: scanFolder - failure to stat '/mnt/playlists/test'
Jun 02 11:05:16 volumio volumio[899]: info: CorePlayQueue::saveQueue
Jun 02 11:05:16 volumio volumio[899]: info: CoreCommandRouter::volumioPushQueue
Jun 02 11:05:16 volumio volumio[899]: info: CoreStateMachine::updateTrackBlock
Jun 02 11:05:16 volumio volumio[899]: info: CorePlayQueue::getTrackBlock
Jun 02 11:05:16 volumio volumio[899]: info: CoreCommandRouter::volumioPlay
```

Can't reproduce for 'Add to Queue' though.
```
Jun 02 11:06:41 volumio volumio[899]: info: Enqueue test
Jun 02 11:06:41 volumio volumio[899]: info: CoreCommandRouter::volumioAddQueueItems
Jun 02 11:06:41 volumio volumio[899]: info: CoreStateMachine::addQueueItems
Jun 02 11:06:41 volumio volumio[899]: info: CorePlayQueue::addQueueItems
Jun 02 11:06:41 volumio volumio[899]: info: Adding Item to queue: mnt/NAS/testbox/ABBA/ABBA_-_The_Complete_Gold_Collection_-_Disc_1/01-Dancing_Queen.flac
.. all the playlist items get added
```

Clear and Play does indeed work ok
```
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::volumioReplaceandPlayItems
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::ClearQueue
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::stop
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::setConsumeUpdateService undefined
Jun 02 11:10:09 volumio volumio[899]: UNSET VOLATILE
Jun 02 11:10:09 volumio volumio[899]: info: CorePlayQueue::clearPlayQueue
Jun 02 11:10:09 volumio volumio[899]: info: CorePlayQueue::saveQueue
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::volumioPushState
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::volumioPushQueue
Jun 02 11:10:09 volumio volumio[899]: info: Play playlist test
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::volumioClearQueue
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::ClearQueue
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::stop
Jun 02 11:10:09 volumio volumio[899]: info: CoreStateMachine::setConsumeUpdateService undefined
Jun 02 11:10:09 volumio volumio[899]: UNSET VOLATILE
Jun 02 11:10:09 volumio volumio[899]: info: CorePlayQueue::clearPlayQueue
Jun 02 11:10:09 volumio volumio[899]: info: CorePlayQueue::saveQueue
Jun 02 11:10:09 volumio volumio[899]: info: CoreCommandRouter::volumioPushState
```

as does Delete
```
Jun 02 11:11:50 volumio volumio[899]: error: error
Jun 02 11:11:50 volumio volumio[899]: error: error
Jun 02 11:11:51 volumio volumio[899]: error: error
Jun 02 11:11:51 volumio volumio[899]: error: error
Jun 02 11:11:51 volumio volumio[899]: info: Deleting playlist test
Jun 02 11:11:51 volumio volumio[899]: info: Listing playlists
Jun 02 11:11:51 volumio volumio[899]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Jun 02 11:11:51 volumio volumio[899]: info: CURURI: playlists
Jun 02 11:11:51 volumio volumio[899]: info: Listing playlists
```

xipmix, 2020-06-02:
The explodeUri routine in app/plugins/music_service/mpd/index.js  seems to force prefixing the uri with '/mnt' in the playlist case

https://github.com/volumio/Volumio2/blob/8583c814bdc88e8c28c10464bec8226bce2b0634/app/plugins/music_service/mpd/index.js#L2260-L2264

There needs to be a case to handle playlist uris (match on 'playlist/.*')

xipmix, 2020-09-14:
This may have been fixed by https://github.com/volumio/Volumio2-UI/commit/09a4ebec9836c4070bab5759c1faa2cc94f49916 .
The trick is that with a playlist you don't want to try to 'play' it directly, you need to replace the queue and play the contents of the playlist.

JeepHu, 2026-02-12:
Sorry for resurrecting such an old thread but I still have the same problem. There's often /mnt as a prefix in /volumio/app/plugins/music_service/mpd/intex.js but where should it be modified, please?