volumio/Volumio2 #1973: Raspberry Pi + Adafruit-MAX98357 - Audio output pops and clicks

timothyjward, 2020-05-20:
The Adafruit-MAX98357A is a cheap but flexible little I2S DAC/Amplifier, and really easy to set up with a Raspberry Pi Zero for a portable smart speaker. Volumio even has [built-in support](https://github.com/volumio/Volumio2/blob/a4e73ab3bab9527bb81808272661ed8bdf4b5586/app/plugins/system_controller/i2s_dacs/dacs.json#L3) for it.

The problem is that the Adafruit-MAX98357A isn't sent any I2S clock until some audio starts, then when the clock suddenly starts this causes a pop/click. This happens *every* time audio starts up, even after pausing a track.

The Volumio support is pretty minimal, and doesn't implement the [full setup suggestions for avoiding pops and clicks](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/raspberry-pi-usage). 

As a user, what I want is for the built-in support to work without pops/clicks.

It is pretty simple to achieve, and I have successfully worked around it:

1. The audio must be routed through dmixer
2. There must be a system service playing from `dev/zero` into dmixer (e.g. using `aplay`)

This ensures that there is *always* an I2S clock (even if it's playing silence most of the time).

I have tried the following minimal `asound.conf`:

    pcm.softvolume {
        type             plug
        slave.pcm       "softvol"
    }

    pcm.softvol {
        type            softvol
        slave {
            pcm         "dmix:1,0"
        }
        control {
            name        "SoftMaster"
            card        1
            device      0
        }
    max_dB 0.0
    min_dB -50.0
    resolution 100
    }

With this service starting at startup:

    /usr/bin/aplay -D dmix:1,0 -t raw -r 48000 -c 2 -f S32_LE /dev/zero

I am aware that there are some limitations with this approach, the most significant of which is that most audio will end up being resampled (dmix runs at 48kHz 32bit by default). I see three reasons that this isn't a significant issue:

1. Audio has to be post-processed for this DAC anyway as it requires software volume control
2. Volumio has a high-quality resampled built in, so all audio can be sent to dmixer with the correct settings - this will avoid any low quality resampling lower in the stack.
3. The Adafruit-MAX98357 is really designed for driving relatively small low-power speakers, not super expensive reference quality drivers where you would be able to tell that resampling had occurred.

A "full featured" solution in Volumio would let the user select what sample rate/depth they wanted to use in dmixer, and it would set up Volumio's resampling engine to convert to that rate.

I don't have much experience with the way in which Volumio works internally, and therefore I have no idea how difficult this would be to achieve. I really would like to stop running a manually patched system though!

nilthing9, 2020-05-22:
Hi, I'm trying to remove the pops / clicks from my volumio setup myself but without much luck. Can you help? I've asked for some help over on the volumio forum, but no one appears to have offered any suggestions.  Could you possibly give me a quick run-through? I tried to follow the instructions over on the adafruit website but couldn't make them stick! My thread is regarding this is [here](https://forum.volumio.org/adafruit-i2s-dac-reducing-pops-crackle-help-needed-t14466.html). I've probably made a mistake somewhere but technical knowledge is a little lacking. Many thanks. Chris

timothyjward, 2020-05-22:
So I went for a much simplified version of the alsa config, as you can see above I just made a very minor modification to the normal "soft volume" that gets created by Volumio.

Specifically I changed the pcm.softvol to target dmixer, using card 1 (the i2s DAC):

    slave {
        pcm         "dmix:1,0"
    }

This is the *only* change I made to the Volumio audio config.

What I also did was to take the service definition from the [Adafruit script](https://github.com/adafruit/Raspberry-Pi-Installer-Scripts/blob/master/i2samp.sh#L445-L454) and tweak it a bit:

The main thing you need to do is to point aplay at the correct target, and use the correct sample rate/bit depth. I used:

    /usr/bin/aplay -D dmix:1,0 -t raw -r 48000 -c 2 -f S32_LE /dev/zero

I chose 48 kHz because that's the default for dmixer, and when I tried to have a custom configuration for it in `alsa.conf` I couldn't make it work. In reality I would prefer 44100 Hz as I'm mostly listening to FLAC CD audio.

So if you set up the systemd service (as per the linked script) but with the updated device target then Volumio should play fine with no clicks at all. For extra quality consider enabling Volumio's resampling and setting its output to 48 kHz/32 bit. That way there won't be a sampling conversion in dmixer using an unknown algorithm).

I hope this helps.

xipmix, 2020-05-23:
Thanks for the detailed report. Can I clarify one thing - is the asound.conf you are currently using the same as you posted in your first comment? If it has changed at all, would you mind reposting it please?

timothyjward, 2020-05-23:
> Thanks for the detailed report. Can I clarify one thing - is the asound.conf you are currently using the same as you posted in your first comment? If it has changed at all, would you mind reposting it please?

Yes, the comment in the first post is 100% of my working `asound.conf`. It represents the minimum necessary change from what Volumio created for me when I enabled software volume control.

I know it's possible to do something more sophisticated and set a different sample rate for dmixer (which would be a great addition as part of the fix) but for this particular DAC/speaker combo it is probably overkill.

The config for my systemd service is:

```
[Unit]
Description=Invoke aplay from /dev/zero at system start.

[Service]
ExecStart=/usr/bin/aplay -D dmix:1,0 -t raw -r 48000 -c 2 -f S32_LE /dev/zero

[Install]
WantedBy=multi-user.target
```

nilthing9, 2020-05-23:
Hi Tim, I managed to get everything to work - working out how to set up my own service caused me to scratch my head for a few minutes. Otherwise - went very smoothly. 
This should definitely be added to the default setup to Volumio. 👍 :) Thanks for your help.

timothyjward, 2020-06-11:
For reference I've managed to get working what should be a (slightly) higher quality setup. It's almost identical to the previous option, but makes use of the [complex example from the alsa docs](https://alsa.opensrc.org/Dmix#6._The_complex_approach_.28defining_dmix_parameters.29). This setup changes the dmixer configuration to use 44.1kHz 16bit audio, which is a perfect match for the majority of my music (FLAC encoded CD audio). I now have the Volumio resampler set to target 44.1kHz 16bit output, which I hope will leave any existing audio untouched if it matches that target.

My `/etc/asound.conf` now looks like this - the `softvolume` feeds into a plug called `dsp` (in case any audio isn't coming in at 16bit 44.1kHz). That plug connects to the `ossmix` dmixer, which is set up with the audio quality I want.

```
pcm.softvolume {
    type             plug
    slave.pcm       "softvol"
}

pcm.softvol {
    type            softvol
    slave {
        pcm         "dsp"
    }
    control {
        name        "SoftMaster"
        card        1
        device      0
    }
max_dB 0.0
min_dB -50.0
resolution 100
}

pcm.ossmix {
    type dmix
    ipc_key 1024 # must be unique!
    ipc_key_add_uid false
    ipc_perm 0666
    slave {
        pcm "hw:1,0"      # you cannot use a "plug" device here, darn.
        period_time 0
        period_size 1024
        buffer_size 8192
        format "S16_LE"
        rate 44100
    }
    bindings {
        0 0
        1 1
    }
}
pcm.dsp {
    type plug
    slave.pcm "ossmix"
}

ctl.mixer0 {
    type hw
    card 0
}
```

The background service playing silent audio is now using the following command to play constant silence into the `ossmix` pcm.

    /usr/bin/aplay -D ossmix -t raw -r 44100 -c 2 -f S16_LE /dev/zero

As of yet I haven't managed to get this setup working with the equaliser plugin, but I will keep digging! 😄

timothyjward, 2020-06-15:
> As of yet I haven't managed to get this setup working with the equaliser plugin, but I will keep digging! 😄

Some further progress (sort of). It turns out that not only does setting this config make the equaliser plugin fail, it also affects a few other things:

* The startup sound doesn't play
* Airplay support (via shairport-sync) doesn't work
* Probably some other stuff

The reason for this stuff not working ~always~ in every case that I've seen is that the relevant audio source/plugin tries to play directly to the hardware device (usually using `plughw:` to allow ALSA resampling if needed). 

I'm relatively new to this, so I may be missing something subtle (or obvious) but this seems like a poor internal design choice, and one where Volumio could improve substantially to allow better plugin control/integration, more consistency internally, and a simpler setup for anything looking to play sound.

It seems to me that a setup like the following would improve matters:

```
# The main Volumio pcm device - all players everywhere send sound here
pcm.volumio {
    type             plug
    slave.pcm       "volumioOut"
}

# The Volumio target hardware output (typically a DAC, but whatever is set in config)
# This is also set as using a plug so that any final resampling transforms needed to
# match the hardware support happen automatically
pcm.volumioOut {
    type            plug
    slave.pcm {
        type            hw
        card            1
    }
}
```

Now rather than having lots of places make a `"plughw:" + card` string they can just use `"volumio"` everywhere. This setup simplifies usage, but still honours Volumio's "bit perfect" audio guarantee. The sorts of places that I'm thinking are `/tmp/shairport-sync.conf` and [where the startup sound gets played].(https://github.com/volumio/Volumio2/blob/a623df3320d0bed262f6d5c26ba269833e68b2a8/app/platformSpecific.js#L88-L97).

The setup also also avoids the need to have custom logic for software volume:

```
pcm.volumio {
    type             plug
    # Volumio configuration inserts "softvolume" into the chain
    slave.pcm       "softvolume"
}

pcm.volumioOut {
    type            plug
    slave.pcm {
        type            hw
        card            1
    }
}
pcm.softvolume {
    type             plug
    slave.pcm       "softvol"
}

pcm.softvol {
    type            softvol
    slave {
        # Volumio configuration tells the last plugin to output to "volumioOut"
        pcm         "volumioOut"
    }
    control {
        name        "SoftMaster"
        card        1
        device      0
    }
    max_dB 0.0
    min_dB -50.0
    resolution 100
}
```

This setup can be extended to include more steps if needed by plugins (e.g. equaliser), where the plugin provides its input pcm name (so that the previous step can point at it) and is given an output pcm name (so that it can point at the next step). An expert UI could also provide a view to reorder plugins (e.g. put the equaliser before the software volume control).