volumio/Volumio2 #1976: Fresh install and no display on 5-inch screen with touch screen plugin installed

jack2675, 2020-05-31:
Hello,
I did a fresh install of Volumio v.2.777 and installes touch screen plugins to have the Volumio UI enabled on my 5-inch display. It seems that installation of touch screen plugins ended with error especially to install chromium due to unmet dependencies... after enabling touch screen plugins no UI is displayed. Could you please help? Please notice that I saw it working ten days ago with the previous version of Volumio and probably the previous plugin which has been updated on 05.25.2020... also, in my case I never had the touch screen running because of the display controller which is XT2046 but only used to have the UI enabled on the screen. BTW thank you for the great job done with Volumio especially when my display worked! Many thanks in advance for your support.

chsims1, 2020-06-02:
This is possibly a result of expired SSL certificates preventing downloads from the Volumio repos.  Hopefully, it should be sorted soon.

jack2675, 2020-06-02:
Thank you chsims1, you mean that a new release is expected?

gvolt, 2020-06-02:
Problem [solved](https://github.com/volumio/volumio-plugins/issues/449#issuecomment-637816827).

jack2675, 2020-06-08:
thanks to all contributors to solve this issue!

closed 2020-06-08 by jack2675