volumio/Volumio2 #1977: Rotary encoder doesn't work after reboot.

MrPi3141, 2020-06-01:
The rotary encoder doesn't work when Voluminous is restarted. I have to go into the plugin settings and click save every time I want to use the encoder.

Saiyato, 2020-06-04:
Hi @MrPi3141 , this has already been addressed and fixed as of version 1.2.4 :)

MrPi3141, 2020-06-11:
Thanks for the feedback. Issue remains unfortunately. We're running the latest version of Volumio on a RPi3B and justboom amp hat. Have tried a couple of encoders, but same issue.  Any help appreciated.

Saiyato, 2020-06-11:
Hi, can you please add some logging? There's debug logging available to explode logging

MrPi3141, 2020-06-11:
Thanks, here's the log from just now. I have marked where the plugin settings were saved which allows the encoder to function. Prior to that, nothing happened when I turned the encoder.

`volumio@volumio:~$ sudo journalctl -f
-- Logs begin at Thu 2020-06-11 19:09:56 UTC. --
Jun 11 19:12:12 volumio volumio[1073]: info: Plugin example_plugin is not enabled
Jun 11 19:12:12 volumio volumio[1073]: info: Loading plugin "inputs"...
Jun 11 19:12:12 volumio volumio[1073]: info: Loading plugin "updater_comm"...
Jun 11 19:12:12 volumio sudo[1292]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Jun 11 19:12:12 volumio sudo[1292]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Jun 11 19:12:12 volumio volumio[1073]: info: Plugin mpdemulation is not enabled
Jun 11 19:12:12 volumio volumio[1073]: info: Loading plugin "rest_api"...
Jun 11 19:12:12 volumio volumio[1073]: info: Loading plugin "websocket"...
Jun 11 19:12:12 volumio volumio[1073]: info: Loading plugin "touch_display"...
Jun 11 19:12:12 volumio volumio[1073]: Forking 3 albumart workers
Jun 11 19:12:13 volumio volumio[1073]: info: Loading plugin "rotaryencoder"...
Jun 11 19:12:13 volumio volumio[1073]: Starting albumart workers
Jun 11 19:12:13 volumio volumio[1073]: Starting albumart workers
Jun 11 19:12:13 volumio volumio[1073]: Starting albumart workers
Jun 11 19:12:14 volumio volumio-remote-updater[639]: [2020-06-11 19:12:14] [connect] Successful connection
Jun 11 19:12:14 volumio volumio[1073]: info: Applying required configuration parameters for plugin rotaryencoder
Jun 11 19:12:14 volumio volumio[1073]: info: ___________ START PLUGINS ___________
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: mpd , getConfigParam
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
Jun 11 19:12:14 volumio volumio[1073]: info: [1591902734341] CoreMusicLibrary::Adding element Media Servers
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jun 11 19:12:14 volumio volumio[1073]: info: Could not detect Primo: Error: Command failed: aplay -l | grep es90x8q2m-dac-dai-0
Jun 11 19:12:14 volumio volumio[1073]: info: Could not detect Primo: Error: Command failed: aplay -l | grep es90x8q2m-dac-dai-0
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
Jun 11 19:12:14 volumio volumio[1073]: info: [1591902734496] CoreMusicLibrary::Adding element Last_100
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
Jun 11 19:12:14 volumio volumio[1073]: info: [1591902734502] CoreMusicLibrary::Adding element Webradio
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: system , getSystemVersion
Jun 11 19:12:14 volumio volumio[1073]: info: [Rotary encoder] Loaded configuration: {"longPressThresholdInMilliseconds":{"type":"int","value":1000},"first_encoder_CLK":{"type":"int","value":14},"first_encoder_DT":{"type":"int","value":15},"first_encoder_SW":{"type":"int","value":0},"first_encoder_encoding":{"type":"int","value":0},"first_encoder_detentActionType":{"type":"int","value":1},"first_encoder_buttonActionType":{"type":"int","value":0},"first_encoder_longPressActionType":{"type":"int","value":0},"second_encoder_CLK":{"type":"int","value":0},"second_encoder_DT":{"type":"int","value":0},"second_encoder_SW":{"type":"int","value":0},"second_encoder_encoding":{"type":"int","value":0},"second_encoder_detentActionType":{"type":"int","value":0},"second_encoder_buttonActionType":{"type":"int","value":0},"second_encoder_longPressActionType":{"type":"int","value":0},"enable_debug_logging":{"type":"boolean","value":true}}
Jun 11 19:12:14 volumio volumio[1073]: info: -------------------------------------------
Jun 11 19:12:14 volumio volumio[1073]: info: -----    MyVolumio plugins startup     ----
Jun 11 19:12:14 volumio volumio[1073]: info: -------------------------------------------
Jun 11 19:12:14 volumio volumio[1073]: info: [MyVolumio PluginManager] Fetching plans data....
Jun 11 19:12:14 volumio sudo[1347]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl start volumio-kiosk.service
Jun 11 19:12:14 volumio volumio[1073]: info: Loading i18n strings for locale en
Jun 11 19:12:14 volumio sudo[1347]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jun 11 19:12:14 volumio volumio[1073]: Updating browse sources language
Jun 11 19:12:14 volumio volumio[1073]: Cannot find translation for sourceMedia Servers
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: my_music , getDisabledSources
Jun 11 19:12:14 volumio systemd[1]: Started Start Volumio Kiosk.
Jun 11 19:12:14 volumio sudo[1347]: pam_unix(sudo:session): session closed for user root
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: mpd , getConfigParam
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::initPlayerControls
Jun 11 19:12:14 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
[... cut ...]
Jun 11 19:13:16 volumio volumio[1073]: info: [Rotary encoder] 1/4 settings loaded
Jun 11 19:13:16 volumio volumio[1073]: info: [Rotary encoder] 2/4 settings loaded
Jun 11 19:13:16 volumio volumio[1073]: info: [Rotary encoder] 3/4 settings loaded
Jun 11 19:13:16 volumio volumio[1073]: info: [Rotary encoder] 4/4 settings loaded
Jun 11 19:13:24 volumio volumio[1073]: info: CALLMETHOD: user_interface rotaryencoder updateGlobalSettings [object Object]
Jun 11 19:13:24 volumio volumio[1073]: info: CoreCommandRouter::executeOnPlugin: rotaryencoder , updateGlobalSettings`

Saiyato, 2020-06-15:
I'm inclined to say you're running v1.2.2, can you config this please? You can see it in overview, if so, you're not running the latest version, which causes your problem (b/c you don't have a switch configured, this is a known and solved bug).

MrPi3141, 2020-06-16:
Thanks, I am running version 2.777, as in the attached photo.
[image]

Saiyato, 2020-06-17:
Hi buddy,

I mean the version of the plugin, you can view that in the plugins page (on the tab "installed plugins")
It should read version 1.2.4 there, otherwise you're a release behind unfortunately.

MrPi3141, 2020-06-17:
Ahh, thanks! It is running version 1.2.2. I have managed to fix the issue by using the advice from https://github.com/Saiyato/volumio-rotary-encoder-plugin/issues/18 - adding a non-existent switch pin to the encoder settings. Thanks for all your help though! I will now close this issue.

closed 2020-06-17 by MrPi3141