volumio/Volumio2 #1979: USB HDD with ext4 seems to be mounted with unsufficient permissions

lbm-services, 2020-06-04:
My HDD could not be scanned unless I changed permissions via ssh.

```
May 25 23:58 : client: [1] opened from 127.0.0.1:54686
May 25 23:58 : exception: Failed to register /var/lib/mpd/music/USB/HDD500: inotify_add_watch() has failed: Permission denied

```
After changing permission to 777, scanning was possible.

xipmix, 2020-06-10:
I think the system will mount the disk and use the permissions in the filesystem on the disk. (It's different if you are mounting an NTFS or FAT32 filesystem, those have options to set the user and permission)

What were the permissions before you changed them?
Does mode 0755 work?
It would be helpful to see the output of
```
$ ls -ld /var/lib/mpd/music/USB/HDD500
```

lbm-services, 2020-06-11:
Sure, fs is ext4. Permission were 755. 
775 works as well.

`drwxrwxr-x 424 volumio volumio 36864 Jun 11 14:25 /var/lib/mpd/music/USB/HDD500/`

xipmix, 2020-06-12:
That's weird, that it can't scan with permissions 0755. If I have time I'll try to reproduce. Can you post your version (cat /etc/os-release will do)

lbm-services, 2020-06-28:
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="b9cd485c2d7861a79d18f42ff0fdb052338b2239"
VOLUMIO_FE_VERSION="2932a2aa65665ab9ec21cd0bdf3d21669bbbf643"
VOLUMIO_BE_VERSION="8583c814bdc88e8c28c10464bec8226bce2b0634"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Mon May 25 18:53:02 CEST 2020"

VOLUMIO_VERSION="2.777"
VOLUMIO_HARDWARE="pi"
VOLUMIO_HASH="c75975235b22f56393d7c4516aa10590"