volumio/Volumio2 #1982: FLAC webradios stop after one song.

loguerto, 2020-06-19:


xipmix, 2020-09-07:
Tried this with motherearth, on version v2.777. It seems to transition just fine, at least the playback display updates as expected.
Here's the transition from one track to the next, this is actually from the second to the third track while I was watching
```
Sep 07 14:19:17 volumio volumio[897]: ---------------------------- MPD announces state update: player
Sep 07 14:19:17 volumio volumio[897]: info: ControllerMpd::getState
Sep 07 14:19:17 volumio volumio[897]: verbose: ControllerMpd::sendMpdCommand status
Sep 07 14:19:17 volumio volumio[897]: info: ------------------------------ 7ms
Sep 07 14:19:17 volumio volumio[897]: verbose: ControllerMpd::parseState
Sep 07 14:19:17 volumio volumio[897]: verbose: ControllerMpd::sendMpdCommand playlistinfo
Sep 07 14:19:17 volumio volumio[897]: verbose: ControllerMpd::parseTrackInfo
Sep 07 14:19:17 volumio volumio[897]: info: ControllerMpd::pushState
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::servicePushState
Sep 07 14:19:17 volumio volumio[897]: info: CorePlayQueue::getTrack 3
Sep 07 14:19:17 volumio volumio[897]: verbose: STATE SERVICE {"status":"play","position":0,"seek":2574330,"duration":0,"samplerate":"96 kHz","bitdepth":"24 bit","channels"
:2,"random":false,"updatedb":false,"repeat":false,"bitrate":null,"isStreaming":false,"title":"The Beatles - Fixing A Hole","artist":"Mother Earth Radio","album":null,"uri"
:"http://icecast3.streamserver24.com:18800/motherearth","trackType":""}
Sep 07 14:19:17 volumio volumio[897]: verbose: CURRENT POSITION 3
Sep 07 14:19:17 volumio volumio[897]: info: CoreStateMachine::syncState   stateService play
Sep 07 14:19:17 volumio volumio[897]: info: CoreStateMachine::syncState   currentStatus play
Sep 07 14:19:17 volumio volumio[897]: info: Received an update from plugin. extracting info from payload
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: albumart , getAlbumArt
Sep 07 14:19:17 volumio volumio[897]: info: CoreStateMachine::pushState
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::volumioPushState
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 07 14:19:17 volumio volumio[897]: info: CoreStateMachine::pushState
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::volumioPushState
Sep 07 14:19:17 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 07 14:19:17 volumio volumio[897]: info: ------------------------------ 32ms
Sep 07 14:19:19 volumio volumio[897]: info:
Sep 07 14:19:19 volumio volumio[897]: ---------------------------- MPD announces system playlist update
Sep 07 14:19:19 volumio volumio[897]: info: Ignoring MPD Status Update
Sep 07 14:19:19 volumio volumio[897]: info:
Sep 07 14:19:19 volumio volumio[897]: ---------------------------- MPD announces state update: player
Sep 07 14:19:19 volumio volumio[897]: info: ControllerMpd::getState
Sep 07 14:19:19 volumio volumio[897]: verbose: ControllerMpd::sendMpdCommand status
Sep 07 14:19:19 volumio volumio[897]: info: ------------------------------ 8ms
Sep 07 14:19:19 volumio volumio[897]: verbose: ControllerMpd::parseState
Sep 07 14:19:19 volumio volumio[897]: verbose: ControllerMpd::sendMpdCommand playlistinfo
Sep 07 14:19:19 volumio volumio[897]: verbose: ControllerMpd::parseTrackInfo
Sep 07 14:19:19 volumio volumio[897]: info: ControllerMpd::pushState
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::servicePushState
Sep 07 14:19:19 volumio volumio[897]: info: CorePlayQueue::getTrack 3
Sep 07 14:19:19 volumio volumio[897]: verbose: STATE SERVICE {"status":"play","position":0,"seek":2576384,"duration":0,"samplerate":"96 kHz","bitdepth":"24 bit","channels":2,"random":false,"updatedb":false,"repeat":false,"bitrate":null,"isStreaming":false,"title":"The Beatles - Fixing A Hole","artist":"Mother Earth Radio","album":null,"uri":"http://icecast3.streamserver24.com:18800/motherearth","trackType":""}
Sep 07 14:19:19 volumio volumio[897]: verbose: CURRENT POSITION 3
Sep 07 14:19:19 volumio volumio[897]: info: CoreStateMachine::syncState   stateService play
Sep 07 14:19:19 volumio volumio[897]: info: CoreStateMachine::syncState   currentStatus play
Sep 07 14:19:19 volumio volumio[897]: info: Received an update from plugin. extracting info from payload
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: albumart , getAlbumArt
Sep 07 14:19:19 volumio volumio[897]: info: CoreStateMachine::pushState
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::volumioPushState
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 07 14:19:19 volumio volumio[897]: info: CoreStateMachine::pushState
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::volumioPushState
Sep 07 14:19:19 volumio volumio[897]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 07 14:19:19 volumio volumio[897]: info: ------------------------------ 45ms
Sep 07 14:19:21 volumio volumio[897]: info:
```
it would really help to have a log from your system of this happening. (https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html)

xipmix, 2020-09-07:
Can't make the other site play, nothing obvious in the log.