volumio/Volumio2 #1985: wpa_supplicant not supplied with country code resulting in inconsistent wifi connections

Rintse, 2020-06-29:
For people outside the US, a country code needs to be set in the configuration of wpa_supplicant to use bands 12-13 (and 14 in JPN). Don't know how feasible this is.

xipmix, 2020-07-01:
This should not be hard to implement.
It looks like all that is needed is
```
country=XX
```
added to wpa_supplicant.conf (where XX is the ISO country code).
See also https://www.npmjs.com/package/country-select-js

This strikes me as related to time zones (https://github.com/volumio/Volumio2/pull/1971)

volumio, 2020-07-01:
We could set it up as "relevant" to the language selected in first config wizard, what do you think

xipmix, 2020-07-01:
So does that mean we'd be using the language code as a proxy for the country code...
Gets tricky quickly, an Italian-speaker in Switzerland for example.

But perhaps you are proposing something else - like an extra selector (specifically for country) in the same dialog?

My thoughts about the UX is that the user wants
 * wireless to work  (so regulatory domain needs to be specified)
 * time to display in local time (99.8% of the time)
 * UI text to be in the language they want

Countries have multiple languages and time zones so to me it seems natural to select the country, then the TZ (from a relevant subset) and the language (from all available options)

ashthespy, 2020-10-05:
How hard would it be to have a "Localisation" tab in the "First start wizard" where you can pick:
1. Language 
2. Country (Prefilled based on 1.)
3. Time zone (Prefilled based on 2.) 

User can always mix and match if they chose, but we shortlist/prefill the most likely choice?

PS: I know this is a bit OT, but https://github.com/volumio/Volumio2-UI/issues/317 was closed for some reason..

xipmix, 2020-10-06:
That sounds fine to me. The prefill could be a second iteration, even.

volumio, 2020-10-06:
I am a fan of simple things... Here we'll have a huge huge list to put on a wizard... See: https://en.wikipedia.org/wiki/List_of_tz_database_time_zones

Ideally we would love to auto select the country from a json list (someone has a link of a good json with Country, Language and Time zone?

timothyjward, 2020-10-06:
> Ideally we would love to auto select the country from a json list (someone has a link of a good json with Country, Language and Time zone?

I assume that the auto-detection would be based on the `Accept-Language` header sent by the browser? That would give you the best guess for language and then possibly also country/timezone if the header value has that information.

ashthespy, 2020-10-06:
You could also use [`navigator.languages `](https://developer.mozilla.org/en-US/docs/Web/API/NavigatorLanguage/languages) but again, this might make things tricky for counties with multiple languages.

ashthespy, 2022-12-28:
FWIW the original issue of setting the wireless domain for `wpa_supplicant` was fixed with https://github.com/volumio/volumio3-os/pull/50