volumio/Volumio2 #1990: Support for MERUS™ Audio Amp piHAT ZW

AMuszkat, 2020-07-10:
Greeting from Infineon Technologies Denmark :)

We have released the [MERUS™ Audio Amp piHAT ZW](https://github.com/Infineon/merus-audio-amp-hat-zw), now with drivers on the Raspberry Pi OS kernel.

We have many customers asking for support of this board on Volumio. 

Could you please indicate how we can get official support in the player?

Best,
Ariel

volumio, 2020-07-10:
Hi Ariel,
glad to read you here! I'm one of the people wanting to try your amp technology and hat! The specs are truly impressive and the fact that one single 2A PSU can drive it is even more impressing.
You can see how to integrate it following this guide:
https://volumio.github.io/docs/I2S_DACs/Adding_Compatibility_to_your_DAC.html

I would be glad also to get in touch with you, you can write me to: michelangelo at volumio dot org

AMuszkat, 2020-07-14:
Hi Michelangelo,

I'm seeing  the following error whe probing the driver:

[   14.490892] snd-rpi-simple soc:sound: ASoC: failed to init link MerusAmp: -517
[   14.492466] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   14.502521] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   14.502832] usbcore: registered new interface driver brcmfmac
[   14.504277] snd-rpi-simple soc:sound: ASoC: failed to init link MerusAmp: -517
[   14.698145] snd-rpi-simple soc:sound: ASoC: failed to init link MerusAmp: -51

Have you seen this with other drivers or do you know what is the fix?

Best,
Ariel

volumio, 2020-07-14:
Sorry, no haven't seen that with other drivers.
Does it work when you start playback or not?

Can you post the output of:
aplay -l
lsmod

AMuszkat, 2020-07-14:
No, the driver is not even loading. We haven't seen this with the 4.19 stable kernel.

volumio@volumioam:~$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
-----------------------------------------------------------------
volumio@volumioam:~$ lsmod
Module                  Size  Used by
bnep                   20480  2 
hci_uart               36864  1 
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             376832  22 hci_uart,bnep,btbcm
ecdh_generic           16384  1 bluetooth
ecc                    32768  1 ecdh_generic
xt_REDIRECT            16384  1 
xt_tcpudp              16384  1 
iptable_nat            16384  1 
nf_nat                 49152  2 iptable_nat,xt_REDIRECT
nf_conntrack          131072  2 xt_REDIRECT,nf_nat
nf_defrag_ipv4         16384  1 nf_conntrack
ip_tables              28672  1 iptable_nat
x_tables               32768  3 ip_tables,xt_tcpudp,xt_REDIRECT
nfsd                  348160  3 
brcmfmac              319488  0 
brcmutil               16384  1 brcmfmac
snd_soc_ma120x0p       16384  0 
regmap_i2c             16384  1 snd_soc_ma120x0p
sha256_generic         16384  0 
libsha256              20480  1 sha256_generic
cfg80211              679936  1 brcmfmac
snd_soc_bcm2835_i2s    16384  0 
snd_soc_rpi_simple_soundcard    16384  0 
rfkill                 28672  4 bluetooth,cfg80211
snd_soc_core          200704  3 snd_soc_bcm2835_i2s,snd_soc_ma120x0p,snd_soc_rpi_simple_soundcard
bcm2835_v4l2           45056  0 
snd_bcm2835            24576  0 
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
bcm2835_codec          36864  0 
v4l2_mem2mem           32768  1 bcm2835_codec
bcm2835_isp            32768  0 
snd_pcm                94208  5 snd_pcm_dmaengine,snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_core,snd_soc_rpi_simple_soundcard
bcm2835_mmal_vchiq     28672  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
snd_seq                69632  0 
videobuf2_vmalloc      16384  1 bcm2835_v4l2
snd_seq_device         16384  1 snd_seq
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
snd_timer              32768  2 snd_seq,snd_pcm
videobuf2_v4l2         28672  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       57344  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
videodev              237568  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
raspberrypi_hwmon      16384  0 
snd                    69632  7 snd_compress,snd_seq,snd_timer,snd_seq_device,snd_bcm2835,snd_soc_core,snd_pcm
mc                     45056  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
i2c_bcm2835            16384  0 
vc_sm_cma              32768  1 bcm2835_mmal_vchiq
uio_pdrv_genirq        16384  0 
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0 
i2c_dev                16384  0 
ipv6                  450560  66 
nf_defrag_ipv6         20480  2 nf_conntrack,ipv6
fuse                  114688  1 
squashfs               40960  1 
overlay               102400  1 
----------------------------------------------------------------------------------------------------
Can not play any audio or access alsamixer for instance

volumio, 2020-07-14:
Did a quick investigation and it seems there was an issue with loading the clocks of drivers:
https://github.com/raspberrypi/linux/issues/3481

It should have been fixed here:
https://github.com/raspberrypi/linux/commit/3dfd41e79a4c00e4d6d801c785209c5346fcd040

I am doing now another build with this updated kernel, will send you link privately ASAP then you can report if that works

AMuszkat, 2020-07-14:
sound great, much appreciated!

AMuszkat, 2020-07-14:
Keep having the same results unfortunately

volumio, 2020-07-14:
That's quite a pain: we made lots of mods to support kernel 5.4, won't be easy to revert them to make this work.

Isn't there anything you can think of on driver side? This should get fixed anyway as kernel will be 5.x from now on

AMuszkat, 2020-07-14:
Let me try to debug between today and tomorrow to see what can I find. Of course, we don't you guys to revert to an older kernel.

volumio, 2020-07-14:
Just in case, I am preparing a build with the latest 4.19 kernel

We might want to do it anyway as there might be other DACs which are affected by this regression in the kernel

AMuszkat, 2020-07-14:
Found the error on linux/sound/soc/bcm/rpi-simple-soundcard.c

codec dai name and I2C adress were swapped. it should be:

SND_SOC_DAILINK_DEFS(merus_amp,
	DAILINK_COMP_ARRAY(COMP_EMPTY()),
	DAILINK_COMP_ARRAY(COMP_CODEC("ma120x0p.1-0020","ma120x0p-amp")),
	DAILINK_COMP_ARRAY(COMP_EMPTY()));

I will make a pull request on linux 5.4

volumio, 2020-07-14:
Ok! Please let me know when its available so we'll do another build.

In the meantime, if you can provide the compiled module, we can add on current build (5.4.51) by overwriting the module

AMuszkat, 2020-07-14:
Already available!

volumio, 2020-07-15:
Yep but it's not available on https://github.com/Hexxeh/rpi-firmware

I am considering making a release with latest 4.19.x kernel...

AMuszkat, 2020-07-15:
ha ok. Let me know if you need the compiled drivers then.

volumio, 2020-07-15:
Ok!

AMuszkat, 2020-07-15:
Here you have the compiled module:

[snd-soc-rpi-simple-soundcard.zip](https://github.com/volumio/Volumio2/files/4924720/snd-soc-rpi-simple-soundcard.zip)

closed 2020-07-22 by AMuszkat

snowshow, 2020-10-31:
> Found the error on linux/sound/soc/bcm/rpi-simple-soundcard.c
> I will make a pull request on linux 5.4

AMuszkat, could you, please, provide us with the link to the working RPI ZW image ?

AMuszkat, 2020-11-01:
Hi,

which image are you referring to specifically? The current driver is working ok on Raspbian with latest kernel.

Best,
Ariel

reynolds17, 2021-03-27:
Hi,

with most recent volumio version the MERUS Audio Amp piHAT ZW does not work. Sound is not playing and Alsamixer  returns error:

cannot load mixer controls: Remote I/O error

dmesg output looks good:
[   18.778233] snd-rpi-simple soc:sound: ASoC: CODEC DAI ma120x0p-amp not registered - will retry
[   19.810127] snd-rpi-simple soc:sound: ASoC: CODEC DAI ma120x0p-amp not registered - will retry
[   19.855491] snd-rpi-simple soc:sound: ASoC: CODEC DAI ma120x0p-amp not registered - will retry
[   22.789945] snd-rpi-simple soc:sound: ma120x0p-amp <-> 20203000.i2s mapping ok

volumio@volumio:~$ cat /etc/*-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="e81b2037a65d8f554cb0c857a1007d659ad7f838"
VOLUMIO_FE_VERSION="e489037accd8e0b2dcbd6e0b054400d8cd36d4e2"
VOLUMIO_BE_VERSION="25875b197c75cd1ac7eb5664d94cdd19afb06985"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Fri Feb 19 17:39:10 CET 2021"

VOLUMIO_VERSION="2.873"
VOLUMIO_HARDWARE="pi"
VOLUMIO_HASH="781d4c9fbf86d562939968306a6a26f4"

Do you have any possible solution?