volumio/Volumio2 #1992: Actual Beta 2.795 crashed all PI4 and Ethernet connected Volumios

DirkStegemeyer, 2020-07-15:
After updating to 2.795 ALL Raspberry 4 devices and all Ethernet connected devices are basically dead.

The PI 3B+ devices on WLAN updated and are OK - the ones mentioned above could not be re-activated, even getting the SD physically into another device. 

Any flag or value set in 2.795 causing to be "offline"? The device itself is not even getting an IP address anymore - same when moving the SD cards -> completely dead. Any clue? Could access the SD card content and provide files in case helpful.

DirkStegemeyer, 2020-07-15:
...moving an SD updated "within" a PI3B+ WLAN into a PI4 Ethernet works with 2.795 - vice versa not

volumio, 2020-07-15:
We've been experimenting with various new kernels this week, and this version was a nogo, you were unlucky to update to this beta release.
Can you please try this new one:
http://updates.volumio.org/pi/volumio/2.798/volumio-2.798-2020-07-15-pi.img.zip

And report back?

DirkStegemeyer, 2020-07-15:
Will give it a try tomorrow - thanks for the light fast reply!

DirkStegemeyer, 2020-07-17:
Works again - thanks!

closed 2020-07-17 by DirkStegemeyer