volumio/Volumio2 #1996: Updating firmware on Volumio running on Pi4?

minihessel, 2020-08-31:
When running Volumio on Pi4, it is very important to have the firmware that fixes the heat issue the Pi4 had in the early days. When you buy a new Pi4, you want to make sure you have that update. How can one do that in Volumio? Can you SSH into terminal and run sudo apt full upgrade? Okay, that will destroy your Volumio installation, but you can just reinstall.

oneismore, 2020-08-31:
You can SSH into the Pi: https://volumio.github.io/docs/User_Manual/SSH.html (remember to first activate via the dev ui by adding /dev to your Volumio URL). Don't know about upgrading - I think Volumio uses a custom compiled Linux distro.

minihessel, 2020-08-31:
But running the SSH then sudo full upgrade will upgrade firmware, even when
running Volumio?

oneismore, 2020-08-31:
I just came across this: https://volumio.github.io/docs/FAQs/Updating_and_Maintenance.html and while you can do that it will likely end with a bricked system. Likely because of their custom Linux distro. So either manually only update the EEPROM itself (and not the Linux installation, see https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md) or first install another OS, do the upgrade etc. (probably the safest option).

minihessel, 2020-08-31:
So install for example raspberry os 32 bit, run sudo full upgrade, then
install Volumio? What’s the best way to check the current firmware via
terminal?

oneismore, 2020-08-31:
You can actually check the firmware from Volumio itself if you install the System Information plugin (under Plugins > Miscellanea).