volumio/Volumio2 #2014: metadata withb webradio

balbuze, 2020-11-07:
Webradio send metadata but it is used, it does not allow to get info for artist or album.
I made a small mod in the plugin 'Radioparadise'. Now when flac stream is used, we can access info for artist and album.
The new display is like that :
[image]
I think it could be applied for all radios sending data. Most of the time it is displayed like that : 
[image]
using split (' -') to separate artist and title and re affect it on the required fields should do the trick.
The name of the radio is then displayed on the left.
It keeps the display used with other sources, local, qobuz etc. So it is homogenous.
What do you think?
(I not able to do that though...)

volumio, 2020-11-09:
We could try, but we shall assume that all webradios shall encode (separated by - ) title, artist and album, which IMHO is not the case...

balbuze, 2020-11-09:
You are right that this not always the case. Some radio don't even send info. But when something is sent, it is often in the form artist - track. Of course, if it not match required info for metadata, it is not different from what we have today.

balbuze, 2020-11-21:
Back with this...
I tested HifiberryOs and it displays album art and metadata from radio. Not in volumio. I do think that it should work just by re-arranging where and how data are displayed
[image]

pierre-wat, 2021-07-26:
Hi ! Do you have any news about this issue ? I'm very interesting about it.

BebeMischa, 2021-12-10:
Is there any movement on this, or did it just got stored in the freezer?