volumio/Volumio2 #2015: No wired network connection with a Gigabyte GA-Z170X-UD5 TH mainboard

janvgils, 2020-11-07:
Volumio Version: 2.834 x86
Hardware: Gigabyte GA-Z170X-UD5 TH
DAC: n.a.

Good evening,

This afternoon I tried to install and configure Volumio on a PC with a Gigabyte mainboard that has an onboard Intel 1Gb network controller.

To my surprise this network controller wasn’t recognized by the Volumio installation.

It was a long time ago that I ran into such an issue when using Volumio/Linux.

My guess is this has to do with the fact that the package “firmware-linux-nonfree” isn’t part of the default installation.

I hope some can help me in the right direction to solve this before we are going to replace hardware.

Thanks in advance

xipmix, 2020-11-09:
What does '/sbin/ifconfig -a' return - just the loopback device?
More detail on the network card would help - can you run 'lspci' to give a specific model number?
Do you have another network card you could plug into the motherboard - that might be a cheap way forward.

janvgils, 2020-11-10:
Thanks for the response I will investigate further, starting if it is possible to get a bash without a network connection and do some debugging like dmesg, cat /var/log/boot/log, lspci etc.

To be continued

janvgils, 2020-11-14:
After installing the Volumio beta version based on Debian buster the onboard network interface is working.
This issue can be closed.

janvgils, 2020-11-14:
This board needs Volumio buster support

closed 2020-11-14 by janvgils