volumio/Volumio2 #2034: Mopidy as drop in replacement for mpd.

rawdlite, 2021-01-03:
I have about 500Gb of local music files. A lot of them being m4a Files.
When scanning the collection mpd (ffmpg?) repeatedly  crashes and I need to restart the mpd service. I doesn't matter whether the files are mounted on USB or accessed via NAS. I could not establish what specific problem ffmpg or mpd or volumio has with certain files.
I tried feeding each subdirectory separately (mpc update) which kinda worked,  but after a restart volumio initiates a rescan, crashes and leaves me with an empty/broken database.

OTOH mopidy has no issues scanning and rescanning all my files and the Mopidy-MPD module provides the full mpd functionality.
Would it be a feasible option to replace mpd by mopidy?
Are there any required mpd features that mopidy does not provide?
Anything I should be aware of?

ashthespy, 2021-01-06:
Might want to take a look at https://github.com/volumio/Volumio2/issues/562

rawdlite, 2021-01-06:
Thank you very much. Seems my search strategy needs improvement. ;-)
I will try a mopidy replacement and report my findings in #562

ashthespy, 2021-01-06:
I'll close this and reopen that so that discussion is in one place :-) Cheers!

closed 2021-01-06 by ashthespy