volumio/Volumio2 #2039: Confusing Player Name Behavior

mcclurec, 2021-01-13:
I have a Homebridge plugin I author called [Homebridge Volumio Speakers](https://github.com/mcclurec/homebridge-volumio-speakers) which is for getting Volumio instances into Apple Homekit. One of the things I do is pull the zone name from the `getMultiRoomDevices` event. I was hoping I could get a littl ebackground on how/why you choose to store those names the way you do.

For example, I named one of my zones `Downstairs Speakers` in the "System -> Player Name" setting. When I reload that setting page, it shows up as `downstairs-speakers`. and when I retrieve it from the API, it comes through as `Downstairs-speakers`.

As a workaround, I have a [piece of code](https://github.com/mcclurec/homebridge-volumio-speakers/blob/master/src/utils.ts#L24-L30) to clean up the `name` value before passing it into Homekit. However, not everyone wants their names title cased. (internationalization, personal preference, whatever)

Why has the `name` field been implemented this way, and would changing it to respect spaces and cases be an easy fix?

mcclurec, 2021-01-13:
Looks like maybe it's for mDNS broadcasting purposes? https://github.com/volumio/Volumio2/blob/52aad0e2315ceb3b7f95677722214c4c72195a14/app/plugins/system_controller/system/index.js#L302

Instead of storing it exclusively in `/etc/hostname` could you store the friendly name separately and save a transformed name for network reasons?

mcclurec, 2021-01-14:
This looks like the code path that results in the single uppercase letter in the API result. 

https://github.com/volumio/Volumio2/blob/52aad0e2315ceb3b7f95677722214c4c72195a14/app/plugins/audio_interface/upnp/index.js#L199-L203

Not sure if your intent was to do something more along the lines of...
```
var name = nameraw.split("-").map(x => x.charAt(0).toUpperCase() + x.slice(1)).join(" ");
```

ashthespy, 2021-01-14:
This is something that looked at a long time ago, lost interest and just hardcoded my PlayerNames. Have a look at #1530 
I would love if we have a  "Pretty Näme" that is used as the general display name, and from which the hostname is derived and displayed (like in the wizard)