volumio/Volumio2 #2043: Pulse is now the PCM/CTL default, this causes problems with Kodi

Saiyato, 2021-01-16:
Hi,

I've been investigating why the Kodi plugin wouldn't show the HATs anymore. I found that in newer images (not sure as of when), PulseAudio is included and is actually the ALSA default.

Question: is that necessary for the Bluetooth implementation(s)? I don't know enough about the sinks, but I seem to be unable to override these values for a particular process (Kodi). Do you have any tips?

From a design perspective, is it advisable to override the pcm/ctl defaults? The plugin overrides them with asound.conf, but now the PulseAudio config overrides them with Pulse:
```
pcm.!default {
    type pulse
    hint {
        show on
        description "Playback/recording through the PulseAudio sound server"
    }
}

ctl.!default {
    type pulse
}
```

Any tips? @balbuze you seem to understand how ALSA works 🙄 I'm just a noob in that area. From what I understand Kodi has a problem with enumerating ALSA devices if their name is "@" and they're not default.

Also, another device (headphones) was added somewhere along the road, I can't find too much info with a quick search, but presumably it's for the Pi4 to enable aux out ( snd_bcm2835.enable_headphones=1 in /boot/config.txt)? This increases the device index, so my "fix" in asound.conf is not future proof, hence my quest to understand how this work to come up with a durable solution.

My asound.conf section
```
#KODI
	defaults.ctl.card ${CTL_CARD_INDEX}
	defaults.pcm.card ${PCM_CARD_INDEX}
#ENDOFKODI
```

balbuze, 2021-01-16:
Hi,
Yes, Dac enumerating  changed... but
As I told you, the alsa pipeline is being reworked for next Volumio.
Each plugin that need a specific configuration for asound would provide a required file. An unique Default output for all. Volumio will take in charge the update for asound.conf
I suggest you to not lost time with the actual arch...
We can help you for that.

Saiyato, 2021-01-17:
Hi @balbuze ,

The problem with Kodi is that it uses system defaults, the only thing I can think of right now is to create a new device and try to make it visible in Kodi (I could investigate using advanced config to override the soundcard). At this point the PulseAudio config is in the way of Kodi (you can only choose HDMI/Aux/Bluetooth out).

timothyjward, 2021-01-18:
Have you tried telling Kodi to use ALSA rather than pulse? It's not the default, and therefore a bit of a pain, but there are instructions here: https://kodi.wiki/view/PulseAudio#Can_I_force_ALSA_even_with_PulseAudio_installed.3F