volumio/Volumio2 #2047: Factory Reset seems not working and broke Volumio installation

digx77, 2021-01-21:
Hi,
today I tried to perform Factory Reset on my latest Volumio version on a Raspberry Pi.
The Factory reset started (deleteUserData has been sent) and I waited for more than 10min; but the Volumio didn't restart automatically.

I tried than to power off and on again manually; but nothing happened.
Finally I needed to reinstall Volumio OS on my SD card.

Could someone check on their side if any issue occurs with latest version about Factory Reset?

Thanks,
Diego

xipmix, 2021-01-26:
Can you post the actual version please, it will help track back through the releases to see where this problem started.

digx77, 2021-01-30:
Hi,
The current version where I experienced this issue is the version 2.861

Bye

digx77, 2021-04-27:
Hi,
any plan to fix this isse. It is very important especially when you have to update Volumio with new version and the message appears like:

"System Software Integrity check failed, manually edited files were detected. System Software Update is not possible under these circumstances. Please execute a Factory Reset or Delete User Data, then retry."

It is very important a Factory Reset will work properly on a device.

xipmix, 2021-04-27:
I'm not sure when someone will have time to look at this, effort seems to be focused on getting 3.x out the door. But I agree this is important.

There is a newer version available (2.882, 2021-0-04); is it possible for you to test this image?
 - download and burn to a spare sd card
 - set up system and make minimal changes; e.g. scan your music collection or create a playlist of webradios
 - factory reset

Can you think what changes you might have made to the code under /volumio?
It is possible to check like so:
```
volumio@volumio$ cd /volumio; git status
```
That should list files that the system thinks have been modified.

It might be useful to make a simple change to one of the code files (eg change some white space in index.js)
and then try to run the factory reset.

If it wedges up again, you might be able to mount the disk image and inspect the logs to see what went wrong. I can probably help with that if you need.

digx77, 2021-04-27:
Hi,
with latest version 2.882 the issue is still there but more logs can be taken now.
I can see below output on the web-socket and the issue seems is on SAMBA daemon:

"Stopping Samba AD DC daemon: samba.\nStopped LSB: start Samba daemons for the AD DC.

LOG:
sent --> ["deleteUserData"," "]

received:

42["LLogProgress",{"message":"info: Command Delete User Data Received\n"}]
42["LLogProgress",{"message":"info: CoreCommandRouter::executeOnPlugin: system , deleteUserData\n"}]
42["LLogProgress",{"message":"info: Created User Data delete file, rebooting\ninfo: ___________ PLUGINS: Run onVolumioReboot Tasks ___________\ninfo: PLUGIN onReboot : networkfs\n"}]
42["LLogProgress",{"message":"volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl reboot\n"}]
42["LLogProgress",{"message":"pam_unix(sudo:session): session opened for user root by (uid=0)\n"}]
42["LLogProgress",{"message":"Started Turns off Raspberry Pi display backlight on shutdown/reboot.\nStarted Synchronise Hardware Clock to System Clock.\nStopping system-ifup.slice.\nRemoved slice system-ifup.slice.\nStopping system-systemd\\x2dfsck.slice.\nRemoved slice system-systemd\\x2dfsck.slice.\nStopping Volumio Tidal Connect Service...\nStopping RealtimeKit Scheduling Policy Service...\nStopping Volumio Bluetooth Module...\nStopping MyVolumio SSH Tunnel...\nreceived signal to exit (15)\nStopping Authenticate and Authorize Users to Run Privileged Tasks...\nStopping Disk Manager (legacy version)...\nStopping Bluetooth service...\nTerminating\nStopping Graphical Interface.\nStopped target Graphical Interface.\nStopping Entropy daemon using the HAVEGE algorithm...\nStopping Multi-User System.\nStopped target Multi-User System.\nStopping mount disks automatically with standby...\nStopping UPnP Renderer front-end to MPD...\nStopping volumio-remote-updater.service...\nStopping Volumio Streaming Daemon...\nStopping Volumio Log Rotation Service...\nStopping OpenBSD Secure Shell server...\nStopping Login Prompts.\nStopped target Login Prompts.\nStopping Getty on tty1...\nStopping Serial Getty on ttyAMA0...\nStopping LSB: start Winbind daemon...\nReceived signal 15; terminating.\nStopping SDP server\nExit\nhaveged: Stopping due to signal 15\nhaveged starting up\nKilling pulse\nStopping LSB: triggerhappy hotkey daemon...\nKilling BT Pair agent\nStopping LSB: start Samba SMB/CIFS daemon (smbd)...\nTerminated\nStopping LSB: start Samba daemons for the AD DC...\nStopping LSB: Start NTP daemon...\nStopping LSB: Brings up/down network automatically...\nvolumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod -R 777 /var/lib/bluetooth/*\n[2021/04/27 14:09:01.497428,  0] ../lib/util/pidfile.c:153(pidfile_unlink)\nFailed to delete pidfile /var/run/samba/smbd.pid. Error was No such file or directory\nStopping LSB: Advanced IEEE 802.11 management daemon...\nStopping LSB: set CPUFreq kernel parameters...\nStopping Shairport Sync - AirPlay Audio Receiver...\nStopping Music Player Daemon...\nStopping D-Bus System Message Bus...\nStarting Store Sound Card State...\nDisconnected from D-Bus, exiting.\nGot SIGTERM, quitting.\nLeaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.135.\navahi-daemon 0.6.31 exiting.\npam_unix(sudo:session): session opened for user root by (uid=0)\nntpd exiting on signal 15\nStopping NTP server: ntpd.\nExiting.\n/bin/chmod: cannot access '/var/lib/bluetooth/*': No such file or directory\npam_unix(sudo:session): session closed for user root\n"}]
42["LLogProgress",{"message":"info: mDNS Browse raised the following error Error: dns service error: unknown\nhaveged.service: main process exited, code=exited, status=143/n/a\nStopped Entropy daemon using the HAVEGE algorithm.\nUnit haveged.service entered failed state.\nStopped mount disks automatically with standby.\nStopped volumio-remote-updater.service.\n"}]
42["LLogProgress",{"message":"Stopped Volumio Log Rotation Service.\nDiscovery: ERRORError: dns service error: unknown\ninfo: mDNS Advertisement raised the following error Error: dns service error: unknown\nStopped Login Service.\nStopped Avahi mDNS/DNS-SD Stack.\nStopped D-Bus System Message Bus.\nStopped Disk Manager (legacy version).\n"}]
42["LLogProgress",{"message":"Stopped Getty on tty1.\nStopped Serial Getty on ttyAMA0.\nStopped Authenticate and Authorize Users to Run Privileged Tasks.\nStopped UPnP Renderer front-end to MPD.\nStopped Shairport Sync - AirPlay Audio Receiver.\nStopped Volumio Streaming Daemon.\nvolumiobt.service: main process exited, code=exited, status=1/FAILURE\nStopped RealtimeKit Scheduling Policy Service.\nStopped Bluetooth service.\n"}]
42["LLogProgress",{"message":"Stopped Volumio Tidal Connect Service.\nStopped MyVolumio SSH Tunnel.\nStopped OpenBSD Secure Shell server.\nNetwork Interface Plugging Daemon...stop eth0...done.\nStopped Volumio Bluetooth Module.\nUnit volumiobt.service entered failed state.\nStopped LSB: triggerhappy hotkey daemon.\nStopped LSB: Start NTP daemon.\nStopped LSB: Brings up/down network automatically.\n"}]
42["LLogProgress",{"message":"Stopped LSB: Advanced IEEE 802.11 management daemon.\nStopped LSB: set CPUFreq kernel parameters.\nStarted Store Sound Card State.\n"}]
42["LLogProgress",{"message":"Stopped Music Player Daemon.\n"}]
42["LLogProgress",{"message":"Stopping SMB/CIFS daemon: smbd.\n"}]
42["LLogProgress",{"message":"Stopped LSB: start Samba SMB/CIFS daemon (smbd).\n"}]
42["LLogProgress",{"message":"Stopping Samba AD DC daemon: samba.\nStopped LSB: start Samba daemons for the AD DC.\n"}]

sngreen, 2022-02-06:
Yes indeed, I upgraded to v 3, noticed both NAS files were not showing up in browser properly, reset to version 2 - at least I thought so - it shows again version 3. Kind of what I would have expected from Volumio - garbage makers!

timothyjward, 2022-02-06:
> Yes indeed, I upgraded to v 3, noticed both NAS files were not showing up in browser properly, reset to version 2 - at least I thought so - it shows again version 3. Kind of what I would have expected from Volumio - garbage makers!

This sounds like a misunderstanding of what a factory reset does. A factory reset will clean the device of all user customisation, effectively taking the device back to a “freshly installed” state. What it won’t (and can’t) do is change the version of Volumio that is running.

 If you have 3.x installed when you start the reset it will be 3.x installed after the reset.

sngreen, 2022-02-06:
I did previously upgrade from 2.xxx to the next 2.xxx releases, whenever available upgrade would pop up in the system settings, and reverting would always put it back to the version the unit was bought with (Volumio Primo). This was my expectation. But fine, it seems like I just need to get used to it.