volumio/Volumio2 #2061: GET on /dev return 404

sk4mi, 2021-01-31:
Hello,

Since update i cannot access to /dev route.

I'm using Volumio 20201213.

```
root@raspberry:~# curl http://volumio.home/dev -v
*   Trying 192.168.1.112...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0xdf18a0)
* Connected to volumio.home (192.168.1.112) port 80 (#0)
> GET /dev HTTP/1.1
> Host: volumio.home
> User-Agent: curl/7.64.0
> Accept: */*
> 
< HTTP/1.1 404 Not Found
< X-Powered-By: Express
< Content-Security-Policy: default-src 'none'
< X-Content-Type-Options: nosniff
< Content-Type: text/html; charset=utf-8
< Content-Length: 142
< Date: Sun, 31 Jan 2021 15:21:40 GMT
< Connection: keep-alive
< 
<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8">
<title>Error</title>
</head>
<body>
<pre>Cannot GET /dev</pre>
</body>
</html>
* Connection #0 to host volumio.home left intact

```

xipmix, 2021-02-04:
Weird, thanks for reporting. Could you please check 2.861 (20201222)?

volumio, 2021-02-04:
MIght be due to this:
https://github.com/volumio/Volumio2/commit/c9e4849495e810bd6ebb6833232daba25d86dd4c

Does it work if you specify an user agent?

sk4mi, 2021-02-08:
As you can see, I already have a user-agent with curl.

Unfortunately I cannot SSH it... default password doesn't works, really weird :/

I think I'm good for a reset..