volumio/Volumio2 #2071: ROON bridge plugin bug

garryhca, 2021-02-08:
The ROON bridge plugin for x86 platform cannot be installed. Here is the temporary fix under SSH. Please add to the source. 

Enable volumio ssh access.
$ curl -O http://download.roonlabs.com/builds/roonserver-installer-linuxx86.sh
$ nano ./roonbridge-installer-linuxx86.sh

Change the PACKAGE_URL= line to the following because Roon official link removed x86 package.
PACKAGE_URL=http://roon-updates-stable.s3.amazonaws.com/RoonBridge_linuxx86.tar.bz2

$ chmod +x roonserver-installer-linuxx86.sh
$ sudo ./roonserver-installer-linuxx86.sh

Reboot. Enjoy!

crieke, 2021-02-08:
I just contacted Roonlabs to fix the link. The error should be fixed there. :-)

garryhca, 2021-02-09:
I think they discontinued x86 version. X86 version is still 1.0. Pi version is 1.7. You should build X64 version of Volumio2 instead of X86.

crieke, 2021-02-09:
they still offer it (with the broken link) on their linux download page:
https://help.roonlabs.com/portal/en/kb/articles/linux-install