volumio/Volumio2 #2097: WebRadio Albumart url is lost when added to Queue

digx77, 2021-03-24:
When browse webradio in webradio section the albumart is shown and the url is correct sent on the websocket:
i.e.: 
{"service":"webradio",
"type":"webradio",
"title":"Absolut Radio",
"artist":"",
"album":"",
**"albumart":"https://radio-directory.firebaseapp.com/volumio/src/images/radio-thumbnails/AbsolutRadio.jpg",**
"uri":"http://stream.absolutradio.de/hq/mp3-160/radioplayer/"}

But when this WebRadio is added to the queue the albumart url is lost and below is reported when in the queue:
{"uri":"http://stream.absolutradio.de/hq/mp3-160/radioplayer/",
	  "title":"Absolut Radio",
	  "service":"webradio",
	  "name":"Absolut Radio",
	  **"albumart":"/albumart",**
	  "samplerate":"",
          "bitdepth":"",
          "channels":0,
          "trackType":"webradio"}

This will result in poor UI in the Queue.

xipmix, 2021-04-27:
This reminds me of issue #1998. Can you please check if it exists on version 2.882 (2021-04-24)? 
If it does it would be really helpful to [send in a log](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html):
 * reboot
 * add webradio to queue
 * capture log

I will have a go at reproducing.

digx77, 2021-04-27:
Hi, just tried with latest version 2.882. Same issue still there: Log also sent with dev UI interface

Starting Live Log...
Reloading.
[/lib/systemd/system/mpd.service:18] Unknown lvalue 'ProtectKernelTunables' in section 'Service'
[/lib/systemd/system/mpd.service:19] Unknown lvalue 'ProtectControlGroups' in section 'Service'
[/lib/systemd/system/mpd.service:20] Unknown lvalue 'ProtectKernelModules' in section 'Service'
[/lib/systemd/system/mpd.service:23] Unknown lvalue 'RestrictNamespaces' in section 'Service'
pam_unix(sudo:session): session closed for user root
info: enable SSH service success
info: CALLMETHOD: system_controller system enableLiveLog true
info: CoreCommandRouter::executeOnPlugin: system , enableLiveLog
info: Launching a new LiveLog session
info: CoreCommandRouter::volumioGetState
info: CorePlayQueue::getTrack 0
info: CoreCommandRouter::volumioGetQueue
info: CoreStateMachine::getQueue
info: CorePlayQueue::getQueue
info: CoreCommandRouter::volumioGetState
info: CorePlayQueue::getTrack 0
info: CoreCommandRouter::volumioGetState
info: CorePlayQueue::getTrack 0
info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
info: CoreCommandRouter::volumioGetState
info: CorePlayQueue::getTrack 0
info: CoreCommandRouter::executeOnPlugin: webradio , handleBrowseUri
info: Getting webradio selection
info: CoreCommandRouter::volumioAddQueueItems
info: CoreStateMachine::addQueueItems
info: CorePlayQueue::addQueueItems
info: Adding Item to queue: http://stream.absolutradio.de/hq/mp3-160/radioplayer/
info: CoreCommandRouter::executeOnPlugin: webradio , explodeUri
info: CorePlayQueue::saveQueue
info: CoreCommandRouter::volumioPushQueue
info: CoreStateMachine::updateTrackBlock
info: CorePlayQueue::getTrackBlock
info: CoreCommandRouter::volumioGetState
info: CorePlayQueue::getTrack 0
info: CoreCommandRouter::volumioGetQueue
info: CoreStateMachine::getQueue
info: CorePlayQueue::getQueue

xipmix, 2021-04-28:
Interesting. I tried this on 2.861 and it worked ok.
 - there was nothing playing and nothing in the queue
 - Go to Webradios->Volumio selection
 - for the first in the list, right-click 'add to queue'
 - I get the radio in the queue and the albumart icon is there too.

I repeated with a second webradio (the second in the list, 'Absolut Relax') and got the same thing, normal behaviour.
Also the queue data seems correct
```
$  /data/queue 
[
  {
    "uri": "http://stream.absolutradio.de/hq/mp3-160/radioplayer/",
    "title": "Absolut Radio",
    "albumart": "https://radio-directory.firebaseapp.com/volumio/src/images/radio-thumbnails/Absolut Radio.jpg",
    "service": "webradio",
    "name": "Absolut Radio",
    "samplerate": "",
    "bitdepth": 0,
    "channels": 0
  },
  {
    "uri": "http://stream.absolutradio.de/relax/mp3-160/radioplayer/",
    "title": "Absolut Relax",
    "albumart": "https://radio-directory.firebaseapp.com/volumio/src/images/radio-thumbnails/Absolut Relax.jpg",
    "service": "webradio",
    "name": "Absolut Relax",
    "samplerate": "",
    "bitdepth": 0,
    "channels": 0
  }
]
```
Can you please try once more like I did (add one or two radios to an empty queue) and paste the contents of ```/data/queue```?
It may be there is an inconsitency with the websocket. How are you checking that, btw? I'm not sure I know how, myself.

xipmix, 2021-04-28:
I did a further check and pressed 'play' on the first item in the queue. It shows the albumart for the webradio in the playback window, after a few seconds delay.

I should add here that I'm using the 'classic' UI with the 'Aurora' background, not sure if that makes any difference.
I switched to 'contemporary' and added another item to the queue (third in the Volumio Selection list), it appeared with its albumart icon ok. So I guess not...

digx77, 2021-04-28:
Hi xipmix,
I found the issue, it is not on Volumio, sorry but on the "addqueue" command sent. The format was old "addqueue" format that works for normal songs while I noticed for Webradio also albumart url need to be passed to Volumio.

Thanks a lot for your check and support. I think this issue can be closed.

xipmix, 2021-04-28:
Great! I believe you can close the issue yourself, no? Please go ahead.

closed 2021-04-28 by digx77