volumio/Volumio2 #2098: moveQueue {from:N,to:N2} command is not working when N (from) is equal to zero

digx77, 2021-03-25:
Using the command moveQueue {from:N,to:N2} it will not work (not move the song) if the song to be moved is at position 0.
So the first song of the queue can never be moved!

Joni-Salminen, 2021-05-06:
I believe the issue is here, and the debug log in volumio/dev kinda confirms it since the error seen in the function is placed in the log if parameters "to" or "from" is equal to zero, so the the if clause evaluates as false.

this could be worked around by null checking the values instead of checking if the values are  >= 1, which propably was not the intention here.

```CoreCommandRouter.prototype.volumioMoveQueue = function (from, to) {
  var defer = libQ.defer();
  this.pushConsoleMessage('CoreCommandRouter::volumioMoveQueue');

  if (from && to) { <-- here 0 == false, so we don't proceed
    return this.stateMachine.moveQueueItem(from, to);
  } else {
    this.logger.error('Cannot move item in queue, from or to parameter missing');
    var queueArray = this.stateMachine.getQueue();
    defer.resolve(queueArray);
    return defer.promise;
  }
};```

digx77, 2021-05-06:
Or just check: if (from>=0 && to>=0)

Joni-Salminen, 2021-05-06:
that does not guard against uninitialised variables or null variables thought, i dont know im not javascript guy. anyway i did make a pull request to get it fixed, hopefully i got correct in there.

digx77, 2021-05-06:
Thanks Joni!

xipmix, 2021-05-18:
FTR the PR is #2116