volumio/Volumio2 #2114: Tidal connect doesn't show up anywhere

gyurobenjamin, 2021-04-30:
I'm running Volumio 2.882 on a RBP 3 A+ with an active Virtuoso subscription.

Everything seems to work apart from tidal connect.

My tidal apps are the latest versions, all my devices are on the same network.

Any suggestion? Can I debug this locally somehow?

mervinderuiter, 2021-04-30:
Hi @gyurobenjamin please take a look at the TidalConnect thread on the community forum. Follow the steps in de faq and let us know if it still doesn't work. 

https://community.volumio.org/t/adding-tidal-connect/42022/220?u=mervinio

gyurobenjamin, 2021-05-07:
@mervinderuiter thanks for the help, this solved my problem: Go to Settings → Sources → Functionalities Control and disable → enable Tidal Connect

Would be great to find this more easily.

closed 2021-05-07 by gyurobenjamin