volumio/Volumio2 #2121: repeatSingle not exposed in API

marciogranzotto, 2021-06-07:
Looks like I can just turn on/off the "repeat all" using:
`volumio.local/api/v1/commands/?cmd=repeat&value=key`

But I don't think there's a way to turn on/off the `repeatSingle` state

Joni-Salminen, 2021-07-10:
You can turn of repeatSingle effectively by turning off repeat, due to how it's handled everywhere..
Thought then when you enable repeat, repeatSingle gets enabled aswell.

I do have pullrequest going on which should handle it, by disabling repeatSingle if repeat is disabled, since those are very tightly coupled in the code.

m4rkireland, 2022-01-13:
Has there been any movement on this issue?

I was thinking of updating the Home Assistant integration but notice that we cannot set `repeatSingle` via the Rest API - is this going to change any time soon?