volumio/Volumio2 #2123: Tidal Connect issue - mytek brooklyn dac+

Zr0-bit, 2021-06-24:
Volumio Information

Volumio Version: 2.882
Hardware: Raspberry Pi 3 B+
DAC: mytek brooklyn dac+

Hello,
first of all thank you for your work!

here my problem:
with the change of the dac does not work tidal connect (previously I had the Pro-Ject Pre Box S2 Digital, now the mytek brooklyn dac+ ). I don’t have any plugins, i switched tidal connect on/off several times, I logged out and logged in from tidal. I tried with shairport activate and deactivate.

In detail:
with Volume Mixer Type: Hardware tidal connect does not appear.
Here the log: [volumioLog_MixerHardware.txt](https://github.com/volumio/Volumio2/files/6708500/volumioLog_MixerHardware.txt)

with Volume Mixer Type: Software tidal connect appear but no audio signal is sent to the dac.
Here the log: [volumioLog_MixerSoftware.txt](https://github.com/volumio/Volumio2/files/6708502/volumioLog_MixerSoftware.txt)

From Volumio app Tidal works with both settings. I also tried every combination related to the volumes of the dac.

thanks for now

mervinderuiter, 2021-07-05:
Hi @Zr0-bit,

We are aware of the problems with the brooklyn dac, and we have made a new test version for soundcard compatability issues. Please follow these instructions and let us know if it works. 

Goto http://{your-volumio-address}/dev and enable test mode. 
Goto http://{your-volumio-address}/plugin/system_controller-system and check for updates. If all goes well version 2.907 can be installed.

Zr0-bit, 2021-07-06:
Hi @mervinderuiter,
 thank you for the reply and for the great work you have done and are doing.

I updated the system to 2.907 and it worked!
I have "undefined" in the Volume Mixer Type menu and sometimes the MQA playing jump to standard 16/44 (not mqa). 

I'll do more tests to check if this error occurs only with tidal connect or also with playback from volumio