volumio/Volumio2 #2136: Volumio on Raspberry Pi Zero W has WiFi-only DHCP issues

oitdmser, 2021-08-07:
Volumio Version: 3.078
Hardware: Raspberry Pi Zero W
DAC: Schiit Modi 3+

As documented [here](https://community.volumio.org/t/pi-zero-w-airplay-wifi-spotty-disconnects/48429), I got volumio running well on the Pi Zero W, went through WiFi setup no problem. It reboots, I see the AirPlay destination and it works great. The next day, it works great. Once in a while, I don’t see the volumio in AirPlay destinations. I can’t ping or SSH into the volumio either. If I wait a while, it’ll come back, or if I go into my WiFi router and force it to reconnect to the Pi Zero the AirPlay destination immediately returns. Note that the RPi is just a few feet of my WiFi AP. I've also done all the troubleshooting on the router side (reboot, update etc.) and that no other devices have issues on my wifi, including other raspberry Pis.

A dump of the logs is [here](http://logs.volumio.org/volumiobuster/V7ZehJr.html) There are multiple entries of `Jul 30 15:43:05 living-room-stereo dnsmasq-dhcp[644]: no address range available for DHCP request via wlan0` one of which was right before the failure.

 Setting a static IP in Volumio's UI fixes the issue, but this is sub-optimal from a UI/UX standpoint.

oitdmser, 2021-12-05:
Some additional testing notes:

1. If I set a static IP in the Volumio UI the problem persists.
2. In years past this has been reported to be an issue with WiFi power management. I can confirm WiFi power management is disabled on this device
3. Writing a script to ping any IP every 5 minutes doesn't solve the issue
4. Others have reported this issue on the community forums, gave up and just connect their devices with an Ethernet adapter
5. The issue also happens on Moode Audio player which also uses Debian on the backend
6. This issue is common with Raspberry Pi boards, apparently their WiFi boards aren't the best

Is it possible to simply write a script that checks WiFi connection status and reloads the wifi if it's not connected? The whole idea of a 'wireless' music player isn't effective if people are required to connect an ethernet cable.

DavidSchmittwilken, 2021-12-10:
I also experienced that issue with my raspberry pi 2 and an additional wifi dongle. So probably the issue is not about the internal wifi, but about the driver. See my bug report from #2096. Unfortunately I didn't find a solution yet. Ethernet also doesn't work for me. I didn't try an ethernet to wifi dongle yet, perhaps that would work. 👯

oitdmser, 2021-12-13:
I don't find many bug reports about Debian's wifi driver for the RPis after 2018 or so, I assumed the driver had been fixed. I know of many people running RPi with Debian and Pi-Hole on wireless without issues.

dasdingo, 2022-01-01:
@oitdmser Last week I installed volumio3 and there I don't have any issues that wifi doesn't reconnect after system going to sleep anymore.