volumio/Volumio2 #2139: Automatic sleep timer to stop music playing after a set time

acjohnson, 2021-08-23:
Looks like someone thought of this idea quite awhile ago but it never received attention.
https://community.volumio.org/t/collected-sleep-timer/2651

Sometimes I'll hit play on an internet radio station on my Volumio Pi and it will just keep playing endlessly. If I forget about it I'll wake up in the morning and find that my covered deck ceiling speakers are still playing the same internet radio station from the day previous.

It would be nice if in the Playback options I could select a duration to automatically stop playback regardless of the audio source. A dropdown with 1, 2, 4, 8, 12, 24 hours for example would be nice. Perhaps even add a check box that would turn off the volumio for those who prefer that functionality.

ashthespy, 2021-08-24:
Hmm does `Settings>Sleep` not work for you?

acjohnson, 2021-08-24:
Not sure how I missed that; yes that works. Is there a way to save it as a persistent setting without requiring you to manually enable sleep each time?

ashthespy, 2021-08-25:
> Is there a way to save it as a persistent setting without requiring you to manually enable sleep each time?

I do not believe so. But you should be able to patch something together as a startup script..