volumio/Volumio2 #2182: Add info about maintenance status.

ashthespy, 2022-01-22:
Disable PRs/Issues and point people towards https://github.com/volumio/volumio3-backend?

volumio, 2022-01-24:
@ashthespy yeah, I totally agree

ashthespy, 2022-01-25:
What is to be done with all the open issues? A lot of them are still valid..

volumio, 2022-01-25:
Yeah but it will take days to get trough them, I would close them IMHO and just open the most urgent onese on volumio3

ashthespy, 2022-01-25:
Are you sure? There are a lot of issues (and fixes/hints/ideas) here that would then be lost. 
I don't think github lets you search issues in two repos easily..

volumio, 2022-01-25:
I would prefer to keep them... but I don't have the time to go trough them, nor the support team..
If you can help by filtering the ones that are more relevant and port them, that would be appreciated

ashthespy, 2022-01-25:
>I would prefer to keep them... but I don't have the time to go trough them, nor the support team..

Then it's a shame that we don't continue to use this repo! Especially considering most of the buster/v3 stuff was done already in this repos https://github.com/volumio/Volumio2/compare/master...buster/master
The codebase alone hasn't changed enough to warrant a whole new repo? More so if issues/PRs are being dropped.  

You could consider [just renaming](https://docs.github.com/en/repositories/creating-and-managing-repositories/renaming-a-repository) this repo to `volumio-core` or what ever.. You keep all your old issues, PRs, forks, and also all the publicity from stars,followers,achievements etc all in one place.
Github will automatically forward links form github.com/volumio/Volumio2 to this new name. 

You also solve the issue of people searching for v2 support/code as `master` would point to well, current code, and v2 could be kept as a `jessie` snapshot.  

> If you can help by filtering the ones that are more relevant and port them, that would be appreciated

Personally, I am not a fan of issue management at $dayjob, let alone in my free time - so not going to be much help. I did close all my open issues though.