volumio/Volumio2 #2183: Play Back Sound With [aplay] at the same time?

xujialiang, 2022-02-06:
When I am playing back musics, at the same time, I try to use play sound with 
` sudo aplay -vv /usr/share/sounds/alsa/Front_Center.wav`

will tell me 

`aplay: main:828: audio open error: Device or resource busy`

how can I do?

timothyjward, 2022-02-06:
> how can I do?

This requires either:
1. A sound card with a hardware mixer
2. A software mixer like dmix

Adding a software mixer can cause a serious drop in audio quality so Volumio doesn’t do it. You could create a custom Volumio plugin to add one, although the general recommendation would be not to do it.

xujialiang, 2022-02-07:
> This requires either:
>
> 1. A sound card with a hardware mixer
> 2. A software mixer like dmix
>
> Adding a software mixer can cause a serious drop in audio quality so Volumio doesn’t do it. You could [...]

Thanks, I want to write a plugin to support sound effect.
My hardware do not support mix.  
So  I try to add software mixer by plugin, but got errors: 
dmix need to connect direct with hw type.
The plugin will always auto create a empty type node. How to create a dmix type node that direct to hw type?

Thanks a lot~